TonyM1958/FoxESS-Cloud #6: illegal timestamp

ondrej337, 2024-02-08:
Hi,
** get_access(), no result data, errno = 40256: Parameter could not be parsed correctly illegal timestamp

Can you check it please?

ondrej337, 2024-02-08:
function: def signed_header(path, login = 0):
    **t_now = time.time()**
    **timestamp = str(round(t_now * 1000))**

ondrej337, 2024-02-08:
Seems foxesscloud company  issue
https://www.foxesscloud.com/public/i18n/en/OpenApiDocument.html#3
your timestamp is correct: 
timestamp | Yes |   | Current timestamp milliseconds

ondrej337, 2024-02-08:
I am UTC+1
Parameter: f.time_zone = "Europe/Prague"
result of the function: ** get_access(), no result data, errno = 40256: Parameter could not be parsed correctly illegal timestamp

I run only:
import foxesscloud.openapi as f

# add your info here
f.api_key = "xxx"
f.device_sn = "xxx"
f.time_zone = "Europe/Prague"

f.get_access_count()

TonyM1958, 2024-02-08:
Ok, can you try increasing the debug level and running again. like this so we can see what the header being sent looks like. This is what it looks like for me:

[image]

TonyM1958, 2024-02-08:
another thing to check is that your system time and time zone are set correctly as time.time() returns a UTC times that relies on the system clock for it's offset.

ondrej337, 2024-02-08:
thank you so much, my time was not correct, so I installed ntp service which synchronize time correctly. You can close the issue.

TonyM1958, 2024-02-08:
OK - glad it's working for you now.

closed 2024-02-08 by TonyM1958