TonyM1958/FoxESS-Cloud #7: Where to store user data (API-key, user, password) if not using Jupyter

JoergZ2, 2024-02-15:
I'm not familar with Jupityer Notebook. I've downloaded foxesscloud from pypi in an virtualenv. Where do I store my access data directly for foxessscloud.com?

JoergZ2, 2024-02-15:
Found how to use without Jupyter

closed 2024-02-15 by JoergZ2

TonyM1958, 2024-02-15:
There are examples of running using bash scripts with python and environment variables here: https://github.com/TonyM1958/FoxESS-Cloud/wiki/Using-Saturn-Cloud

Depends what you want to do and how you want to run different functions.

JoergZ2, 2024-02-15:
Thank you! In the meantime I was playing around a bit with an own python script to get the variables from foxesscloud. I fetched 109 variables. Is there any glossar which are the exact meanings of the variables.  I seem to have found the variables for the current consumption and output of the PV system. But I'm not sure about the values for the battery. What is the input, what is the output. How much power is fed into the grid. Is there a list somewhere with explanations of the terms? And thank you for your great script!