TonyM1958/FoxESS-Cloud #9: Save plots to file

stevetrease, 2024-03-09:
Hi Tony,

Great library. Am using it to update PVOutput and set saving session export schedules and am beginning to explorer the forecasting functionality.

When I use the library in a Jupyter notebook plots are shown inline. Great.
However, I usually run the library from a cronjob and the plots aren't displayed ;-)

It would be great if there were an option to save plots to files so that they can be viewed after the script have finished.

TonyM1958, 2024-03-09:
I get the same when I run the scripts in Saturn Cloud. The matplotlib module can't output when the output device the script is using only supports ASCII text - fortunately, the plot lib doesn't throw an exception. It just throws away the output is would have generated.

There's another problem that cloud hosting environments like Google Colab and Saturn Cloud don't really have anywhere to save runtime files persistently, so it's likely that any file the script writes won't be there.

I'll have a look but I don't know if there is a good alternative to this...

TonyM1958, 2024-03-09:
This looks promising though: https://stackoverflow.com/questions/9622163/save-plot-to-image-file-instead-of-displaying-it

Presumably, you can pass in the name of a file to save the plot to?

TonyM1958, 2024-03-09:
[image]

This look like what you want?

stevetrease, 2024-03-09:
That’s the sort of thing I’m after.

Changing plt.show() to plt.savefig(filename) in the library should work, although that would then stop the inline charts appearing in interactive workbooks.

TonyM1958, 2024-03-09:
All done, just testing, I'll try to upload some code later. Have a look at the README.MD later

stevetrease, 2024-03-10:
Tony, Thank you very much for doing that so quickly. Works perfectly. 
If you are interested, I'm going to use it to push tomorrow's forecast and chart to my iOS devices with Pushover.
Cheers, Steve

TonyM1958, 2024-03-11:
Thanks for the feedback. I've had a look at pushover and created an application key for charge_needed and will look at building push notifications into the code. That should mean I get the charge and chart sent to me each night instead of me remembering to look at it - great addition by the look of it.

Hopefully, you will soon be able to do this just by adding your pushover user key as a setting to your cronjob?

TonyM1958, 2024-03-11:
I've now added f.pushover_user_key to version 2.1.3. If you set this to your user key, set_tariff(), charge_needed() and battery_info() will push a message each time they run. For set_tariff() is just does this if you set 'agile' as the tariff, so you get the up to date pricing for the charge time is has set.

Please give this a go when you get a chance and let me know if it works for you. I've split across 3 different applications keys to get the most of the app notification limit of 10,000 calls pa.

There's still work to do adding some setup instructions around this for users to run in Saturn Cloud.

Thanks for the heads-up!

closed 2024-03-19 by TonyM1958