TonyM1958/FoxESS-Cloud #13: Dateparse issue?

ramotswa, 2024-06-17:
Just opening up my unchanged notebook again and trying to run a simple get_device(). Getting an error.
Code; 

```
import foxesscloud.openapi as f
f.api_key = "valid - checked the portal"
f.device_sn = "valid"
f.time_zone = "Europe/London"
f.get_device()

** get_device(), no list result data, errno = 40256: Parameter could not be parsed correctly illegal timestamp

TonyM1958, 2024-06-17:
That error is produced when Open API takes longer than 60 seconds to process a query. It has been running very slowly this time of day recently. Try again later, after 10pm or tomorrow before 12 noon and it should work.

ramotswa, 2024-06-18:
Yes, quite right. All working fine now. Thanks

closed 2024-06-18 by ramotswa