TonyM1958/FoxESS-Cloud #15: Foxess API servers stopped working completely?

simonf2004, 2024-08-02:
Hello,
Since yesterday I have been getting:
b'upstream connect error or disconnect/reset before headers. retried and the latest reset reason: connection failure' 
No matter which API endpoint I call or from what machine.
Am I the only one, is anyone else experiencing the same thing?
Does this mean that the Foxess API server is completely overloaded?

TonyM1958, 2024-08-02:
I'm getting 503 Service Unavailable so I think Fox has taken the servers down for maintenance / update today. Hopefully, it will come back and be better...

TonyM1958, 2024-08-02:
The Fox private API is also giving operation time outs for the last few minutes.

simonf2004, 2024-08-02:
Thank you sir, lets hope that it will comeback better than before and be finally reliable

TonyM1958, 2024-08-03:
Worth having another look when you get time. The Open API servers are back and, first impression, is that the response time is much improved. Could still revert back to not working of course.

One problem I noted in testing is that you cannot currently set a schedule that includes any time periods with 'ForceCharge' work mode on H1 inverters with firmware 1.74 - the error returned is 'write failure'. I don't know of other inverter models or other firmware versions are affected.

Trying to set any time period that includes 'maxsoc' fails with 'Parameter does not meet expectations'. This problem is also seen when using the Fox web site and Apps that try to set 'maxsoc'.

closed 2024-08-18 by TonyM1958