TonyM1958/FoxESS-Cloud #16: get_schedule returning not supported

cdb27, 2024-09-17:
Hello,

I think this is a bug on the Fox API rather than your code when I call /device/detail?sn={{sn}} then it returns:

[image]

But if I call /device/scheduler/get/flag it returns:

[image]

Seems to be a discrepancy in the Fox API. 

At the moment I can't set schedules using the code as it thinks my device is unsupported because the get_device() returns False on the Scheduler. If I comment out this code from openapi.py from get_flag(), then the code works fine and sets the schedules. 

`if device.get('function') is None or device['function'].get('scheduler') is None or device['function']['scheduler'] == False:
        output(f"** get_schedule() schedules are not supported")
        return None`

My device is a H1-3.7-E-G2.
Master 1.27
Slave 1.02
ARM 1.44

TonyM1958, 2024-09-17:
Thanks for letting me - there are a few discrepancies like this and I can only run my tests against my own inverter model. On my H1-6.0-E, it returns True for both flags...

I'll update the code in Open API v2.5.1 to remove the extra check and also call get_flag() as part of get_device() and update the 'scheduler' flag in the device data. As a work around, you could try adding this to your code after get_device() and before trying to get or set a schedule:

```
f.device['function']['scheduler'] = True
```

There's also conflicting data in the API around whether the inverter supports setting Max Soc in schedules - currently, this is something that works via the Fox API but not via Open API.

TonyM1958, 2024-09-17:
Reviewed the code and results and also found some discrepancies between where Fox uses True/False for enable and also 1/0. I've made some updates so you use either when setting schedules but the code will always return True/False for the enable flag.

closed 2024-10-13 by TonyM1958