TonyM1958/FoxESS-Cloud #17: battery.ipynb   v1.7.3 error

Daibutt, 2024-10-12:
Hi Tony, as requested;

HV BMS v1    with f/w 1.020
8x HV2600 v1   with f/w 1.06

Produces following error;
---------------------------------------------------------------------------

ValueError                                Traceback (most recent call last)

[<ipython-input-2-6e8902cd3f4a>](https://localhost:8080/#) in <cell line: 2>()
      1 # Run this cell to get information about your battery
----> 2 f.battery_info()

1 frames

[/usr/local/lib/python3.10/dist-packages/foxesscloud/foxesscloud.py](https://localhost:8080/#) in get_batteries(info)
    682         if b.get('info') is not None and b['info']['masterVersion'] >= '1.014':
    683             residual_handling = 2
--> 684         capacity = b['ratedCapacity'] / 1000 * int(b['soh']) / 100
    685         soc = b.get('soc')
    686         residual = capacity * soc / 100 if capacity is not None and soc is not None else capacity

ValueError: invalid literal for int() with base 10: '--'

TonyM1958, 2024-10-12:
Thanks. The error is because the v1 BMS is returning an invalid SoH but it would also be useful to know what other correct / incorrect info is being returned.

Please can you run a cell with this, after the error and screen shot the results:

```
f.batteries
```

I've made some updates on v1.7.4 but it is still possible that the v1 BMS is reporting different values for 'capacity' and 'residual'.

Daibutt, 2024-10-12:
Thanks Tony, I'm now getting a different error. 
I'm guressing this is 1.7.4 but not sure as when I run the initial cell it doesn't show the version number below the cell after it's run (it definitely did with 1.7.3).
Anyway this is the new error I get with the f.battery_info cell
[image]

But I can still run f.batteries and this is the result;
[image]

TonyM1958, 2024-10-12:
The token is invalid because it has expired. You need to restart the kernel and run cell 1 again. Under Runtime in the menu, select Disconnect and delete runtime. Then select Run All and it will reload the library and regenerate the token. You will get a different result.

However, from looking at the data, you have there, it looks like I may need to do some more updates so it calculates the capacity correctly.

Daibutt, 2024-10-12:
Brilliant, yes it does now work. But I notice that residual/capacity don't show what I was expecting.
Capacity is showing  19.27kWh - I would expect 20.48kWh (8* 2.56) and 19.27 kWh is what register 11037 is showing (BMS kWh remaining). 

----------------------- BMS 1 -----------------------
SN 67BBHV1xxxxx, HV2600, Version 1.020 (BMS)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 1)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 2)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 3)
SN 602H262xxxxx, HV2600, Version 1.06 (Battery 4)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 5)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 6)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 7)
SN 672H262xxxxx, HV2600, Version 1.06 (Battery 8)

Current SoC:         96%
Capacity:            19.27kWh                                    <<<<<<<<<
Residual:            18.50kWh (SoC x Capacity)         <<<<<<<<<
InvBatVolt:          423.7V
InvBatCurrent:       0.0A
State:               Discharging (0.000kW)
Battery Count:       8 batteries with 16 cells each
Battery Volts:       423.9V total, 52.98V average, 52.99V maximum, 52.96V minimum
Cell Volts:          3.311V average, 3.315V maximum, 3.307V minimum
Cell Imbalance:      0.24%:
BMS Temperature:     23.8°C
BMS Charge Rate:     10.0A (estimated)
Battery Temperature: 21.9°C average, 25.0°C maximum, 20.0°C minimum

Info by battery:
  Battery 1: 52.99V, Cell Imbalance = 0.09%, Average Cell Temperature = 22.9°C
  Battery 2: 52.99V, Cell Imbalance = 0.12%, Average Cell Temperature = 22.4°C
  Battery 3: 52.97V, Cell Imbalance = 0.09%, Average Cell Temperature = 21.6°C
  Battery 4: 52.97V, Cell Imbalance = 0.03%, Average Cell Temperature = 21.2°C
  Battery 5: 52.96V, Cell Imbalance = 0.15%, Average Cell Temperature = 22.1°C
  Battery 6: 52.99V, Cell Imbalance = 0.06%, Average Cell Temperature = 21.9°C
  Battery 7: 52.98V, Cell Imbalance = 0.03%, Average Cell Temperature = 21.2°C
  Battery 8: 52.99V, Cell Imbalance = 0.18%, Average Cell Temperature = 21.5°C

TonyM1958, 2024-10-12:
Yes, that's what I expected would be wrong when I saw your data. Fox is returning a ratedCapacity of 20.48kWh. For the v1 BMS, the Residual returned is the Residual Energy but as you have BMS firmware 1.020, it is expecting this to be Residual Capacity (which is what BMS v2 returns).

With a Residual of 19.27kWH at 96% SoC, the capacity should be 20.07kWh. That gives you an SoH of 98%. To me, that sounds high unless your batteries are only a few months old?

Basically, Fox really don't do a great job of tracking the actual battery capacity in BMS v1 or v2.

TonyM1958, 2024-10-12:
I've pushed v1.7.5. Hopefully, this will resolve the capacity and residual as it now differentiates between BMS v1 and v2 based on the BMS serial number. There's a good chance it will still get this wrong for ECS as I don't have specific examples of what the newer firmware returns for residual yet.

I've added an estimate of SoH based on the current capacity and rated capacity. I don't think the Fox report capacity accurately (certainly for BMS v2 in my experience) so treat this as indicative of where SoH might be, based on the data Fox does provide.

TonyM1958, 2024-10-13:
Turns out ECS returns a value of 0 for ratedCapacity and causes a divide by zero error when calculating SoH.

I've pushed another update (1.76) that should fix this. Please can you give it a spin when you get a chance and confirm that this still works for you? Thanks.

Daibutt, 2024-10-13:
Thanks Tony, yes 1.7.6 works great :)

Current SoC:         96%
Capacity:            19.77kWh (Residual / SoC x 100)
Residual:            18.98kWh
Rated Capacity:      20.48kWh
SoH:                 96.5% (Capacity / Rated Capacity x 100)
InvBatVolt:          423.5V
InvBatCurrent:       0.0A
State:               Charging (0.089kW)
Battery Count:       8 batteries with 16 cells each
Battery Volts:       423.8V total, 52.97V average, 52.98V maximum, 52.96V minimum
Cell Volts:          3.311V average, 3.312V maximum, 3.309V minimum
Cell Imbalance:      0.09%:
BMS Temperature:     23.4°C
BMS Charge Rate:     15.0A (estimated)
Battery Temperature: 20.8°C average, 25.0°C maximum, 19.0°C minimum

Info by battery:
  Battery 1: 52.97V, Cell Imbalance = 0.09%, Average Cell Temperature = 21.4°C
  Battery 2: 52.97V, Cell Imbalance = 0.09%, Average Cell Temperature = 21.0°C
  Battery 3: 52.97V, Cell Imbalance = 0.09%, Average Cell Temperature = 19.9°C
  Battery 4: 52.96V, Cell Imbalance = 0.06%, Average Cell Temperature = 21.0°C
  Battery 5: 52.97V, Cell Imbalance = 0.09%, Average Cell Temperature = 20.9°C
  Battery 6: 52.98V, Cell Imbalance = 0.06%, Average Cell Temperature = 20.1°C
  Battery 7: 52.98V, Cell Imbalance = 0.06%, Average Cell Temperature = 19.9°C
  Battery 8: 52.97V, Cell Imbalance = 0.06%, Average Cell Temperature = 22.0°C

closed 2024-10-14 by Daibutt