TonyM1958/FoxESS-Cloud #18: Battery.ipynb no longer reporting

Andysmith01a, 2024-10-12:
As described on Facebook foxess owners page the app tries to load soh which my V1 ECS batteries don't have the capability to report. My batteries are 4x ECS 2900 BMS is version 1.017 slaves are all version 1.12 the app was working fine until the other day.

Andysmith01a, 2024-10-12:
Please ignore above issue, apparently is now working properly again tvm

TonyM1958, 2024-10-12:
Thanks.

I pushed an update eariler and another one just now. Although it no longer crashes, there is a chance it does not report the battery capacity or residual correctly. I've also added an estimated SoH based on the data Fox returns.

If you can, please can you run the script again (using v1.7.5) and see what it reports. Please can you also run this additional cell below and paste a screen shot of the contents. I'm particularly interested to see the first 7 characters of the BMS serial number and the BMS firmware version as this will help to identify the correct residual handling.

```
f.batteries
```

Andysmith01a, 2024-10-12:
V1.7.5 now not reporting anything again. f.batteries script reports as follows 
[image]
[image]

Andysmith01a, 2024-10-12:
V1.7,5 reports as follows 
[image]

Andysmith01a, 2024-10-12:
[image]

Andysmith01a, 2024-10-12:
[image]

TonyM1958, 2024-10-12:
OK, thanks for posting the info. Clearly, the problem here is that Fox are returning the ratedCapacity of your battery as 0. HV2600 BMS v1 and v2 both return a value for that. I can check for this to stop the error.

Out of interest, what do you see when you look at the battery at foxesscloud.com? The rated capacity is shown as 'Nominal Capacity' here:

[image]

It's also shown in Fox App v2 if you tap the Battery, select Detailed Parameters and then look at Capacity.

TonyM1958, 2024-10-13:
I've pushed v1.7.6. Hopefully, this will resolve the problem of ratedCapacity being returned as 0 and causing the divide by zero error. Please give this a go and confirm if it now works you when you get a chance.

closed 2024-11-02 by TonyM1958