TonyM1958/FoxESS-Cloud #19: Battery capacity double counted

FencingFanatic, 2024-10-30:
I noticed my reported battery capacity when using the charge calculation via Saturn Cloud has recently been doubled up. I have 2 mira hv 25's so slightly under 5kwh but yesterday's is showing my capacity as 9kwh, it is then using this figure in its calculations. Not sure of the outcome of this but I thought it worth mentioning. It's been doing it for a few days I think.

mkdir -p /home/jovyan/setup-logs
sudo chown jovyan:jovyan /home/jovyan
sudo chmod 744 /home/jovyan
mkdir -p /home/jovyan/.conda
mkdir -p /home/jovyan/.conda/pkgs
mkdir -p /home/jovyan/.local/bin
mkdir -p /home/jovyan/.ssh
touch /home/jovyan/.conda/environments.txt
git config --global user.email
git config --global user.name
cp /home/jovyan/.saturn/authorized_keys /home/jovyan/.ssh/authorized_keys
cp /home/jovyan/.saturn/known_hosts /home/jovyan/.ssh/known_hosts
chmod 600 /home/jovyan/.ssh/known_hosts
cp /home/jovyan/.saturn/ssh_config /home/jovyan/.ssh/config
chmod 600 /home/jovyan/.ssh/config
chmod 700 /home/jovyan/.ssh
mkdir -p /home/jovyan/project
chmod 600 /home/jovyan/.ssh/authorized_keys
sleeping
INSTALL PACKAGES
=================================================
Running pip install
Collecting foxesscloud
  Obtaining dependency information for foxesscloud from https://files.pythonhosted.org/packages/0b/95/61569c2660918f229d454c3068d2d55abc2043c2107ed56bef56946aa632/foxesscloud-2.6.5-py3-none-any.whl.metadata
  Downloading foxesscloud-2.6.5-py3-none-any.whl.metadata (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.8/58.8 kB 2.8 MB/s eta 0:00:00
Downloading foxesscloud-2.6.5-py3-none-any.whl (109 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 109.3/109.3 kB 4.5 MB/s eta 0:00:00
Installing collected packages: foxesscloud
Successfully installed foxesscloud-2.6.5
START SCRIPT
=================================================
RUNNING
=================================================
FoxESS-Cloud Open API version 2.6.5
---------------- set_tariff -----------------
Product: AGILE-FLEX-22-11-25
Region:  Eastern England
Price p/kWh inc VAT on 2024-10-29:
  21:00  21.4  18.8  19.0  17.6  19.8  19.9
  00:00  18.6  19.0  18.7  18.6  18.3  17.6
  03:00  18.4  18.2  18.4  15.9  19.0  18.7
  06:00  20.1  21.8  20.7  26.0  21.8  23.6
  09:00  22.5  21.4  21.8  20.1  19.5  19.9
  12:00  21.2  21.6  22.8  21.8  24.0  20.3
  15:00  25.5  27.8  42.2  49.3  46.1  48.5
  18:00  43.4  42.5  22.9  22.0  21.4  19.0
  21:00  21.2  21.2  21.8  19.2
Charge times
  00:00-06:00 at 18.3p
  12:00-16:00 at 23.1p
Tariff set to Agile Octopus
---------------- charge_needed ----------------
Battery Info:
  Capacity:    9.00kWh
  Residual:    2.34kWh
  Voltage:     104.0V
  Current:     2.7A
  State:       Discharging (0.289kW)
  Min SoC:     10% (0.90kWh)
  Current SoC: 26.0%
  Max SoC:     100% (9.00kWh)
  Temperature: 23.7°C
  Max Charge:  15.0A
  Resistance:  0.14 ohms
  Nominal OCV: 105.8V at 60% SoC
** get_remote_settings(), no result data, errno = 40257: Parameter does not meet expectations
Device Info:
  Model:     H1-5.0-E
  Rating:    5.00kW
  Export:    5.00kW
  Charge:    15.0A, 1.62kW, 96.3% efficient
  Discharge: 35.0A, 3.70kW, 97.0% efficient
  Inverter:  101W power consumption
  BMS:       50W power consumption
Consumption (kWh):
  2024-10-27: 13.2, 2024-10-28:  7.7, 2024-10-29:  7.3
  Average of last 3 days: 9.7kWh
Solcast: 2024-10-30 8.9kWh
Generation (kWh):
  2024-10-27: 12.5, 2024-10-28:  4.1, 2024-10-29:  5.7
  Average of last 3 days: 7.4kWh
Charge needed: 4.79kWh (03:02)
  SoC now:     26% at 21:04 on 2024-10-29
  Start SoC:   10% at 01:30 (0.91kWh)
  Charge to:   71% 01:30-05:00 at 17.92p (5.52kWh)
  End SoC:     70% at 06:00 (6.27kWh)
  Contingency: 11% SoC (0.97kWh)
  PV cover:    92% (8.9/9.7)
Setting time periods:
   Time Period 1 = 00:00-00:00
   Time Period 2 = 01:30-05:00 Charge from grid
Battery SoC:
  21:00   26%   21%   18%
  00:00   15%   12%   19%   36%   54%   71%
  06:00   70%   67%   63%   66%   70%   81%
  12:00   93%  100%  100%  100%   93%   81%
  18:00   70%   59%   49%   43%   38%   34%
  00:00   32%   29%   25%   20%   17%   16%
  06:00   14%
------------ set_pvoutput (100993) -------------
Date range 2024-10-28 to 2024-10-29 has 2 days
------------------------------------------------
20241028,4172,1000,3163,10:42,,,,,0,0,6299,0,7699,,,,  # uploaded OK
20241029,5826,2500,2463,12:02,,,,,0,0,5800,0,7300,,,,  # uploaded OK

TonyM1958, 2024-10-30:
Ok. Will have a look. Please can you tell me what BMS firmware you currently have? This may be to do with changes Fox made to reporting of Residual in a recent firmware update.

FencingFanatic, 2024-10-30:
Firmware versions:
Manager 1.74
Slave       1.02
Master    1.65

TonyM1958, 2024-10-30:
That's the inverter firmware. To get the BMS firmware, tap the battery in App v2, tap the serial number and it should show the BMS version. For example, my BMS is v1.014.

You should also be able to work around the problem and over-ride the capacity calculation by adding the parameter capacity=4.96 to charge_needed()

FencingFanatic, 2024-10-30:
Sorry my mistake, Version_BMS 1.016. I'll add the variable as suggested.

TonyM1958, 2024-10-30:
Ok, with that BMS firmware, Fox changed Residual Energy to Residual Capacity and thus throws out the capacity calculation. There is a variable that allows you to add this line to the script before it calls charge_needed():

f.residual_handling=2

That should restore the correct capacity calculation if you want to give it a try instead of adding the capacity parameter.

There is an error **remote_settings in the middle of the log you posted so I will have a look and see if I can work out what this is as well. I suspect this may be where it is trying to check your BMS version to automatically set residual_handling.

If you are using Solcast, you might run out of API calls while testing theses changes. You can over-ride the forecast with another parameter forecast=8.5, for example.

TonyM1958, 2024-10-30:
Please can you let me know the first 8 characters of your BMS serial number? I use tthis to detect the type of BMS and distinguish between HV v1 and v2. Hopefully, I can then also detect the Mira BMS correctly. For example, 60BBHV1... is HV BMS v1 and 60BBHV2... is HV BMS v2

FencingFanatic, 2024-10-30:
The first 8 characters are 60MBB010.

TonyM1958, 2024-10-30:
Thanks. Hopefully, you have a work around for now? It will take a few days to review and update the code. I'll let you know when I make any changes.

FencingFanatic, 2024-10-30:
I tried the f.residual_handling=2 and this time my capacity shows as 2.34kWh. Thanks for looking at the code.

TonyM1958, 2024-10-30:
Ok, that's not right either so hopefully, over riding with capacity=4.96 for now will work better.

I may need you run some diagnostic code at some point to see what is being returned by Fox for the battery. Drop me a message on Messenger when it's convenient and we can talk through the best way to do this, if that's ok.

FencingFanatic, 2024-10-30:
The capacity value seems to have solved it for now, thanks. I'll message you in the next few days.

TonyM1958, 2024-11-01:
I found the error with get_remote_settings. It was trying to get the current work mode but that isn't available via Open API. I'll fix it so the error doesn't get displayed in the next release.

It is looking like Mira BMS is behaving differently to others. My theory is that it is returning residual capacity as 2.34kWh, similar to HV. The difference may be that Mira is returning the current capacity per battery where HV returns the current capacity for all batteries. That would mean your current capacity is 4.68kWh out of a rated capacity of 4.96kWh, giving an SoH of 94.4%. This is about what I would expect for batteries are between 1 and 2 years old.

Charge_needed() tries to work out the current capacity rather than the rated capacity to make sure the charge and discharge, min and max SOC are accurate. You may find it's more accurate if you set capacity=4.68.

I've also found that, with later BMS firmware, the battery charge and discharge losses can vary quite a lot. This is the ratio of the amount of charge the BMS says it has increases compared to amount of charge energy that is applied and conversely, the amount the battery charge drops compared to the battery discharge. Currently, with my HV BMS, that charge loss is 108% and the discharge loss is 95%  (i.e. the battery adds 108% of the kWh charge it gets but the kWh discharge is 95% of what the battery losses). This is another reason I try to detect the BMS and firmware, so I can apply the correct calibration.

FencingFanatic, 2024-11-01:
Thanks for investigating, I shall update the capacity value for accuracy.

TonyM1958, 2024-11-01:
I've updated the released code to v2.6.6. This should load automatically when it is next run. Please let me know if this update causes any problems.

Amongst other things, the update should now work if you add the following line before calling charge_needed() instead of adding the capacity parameter:

```
 f.residual_handling=3
```

The advantage of this is that it will use the residual capacity per battery returned by Fox to work out the current battery capacity and this will track the reduction over time in SoH. This change is a bit experimental so it may or may not work. If it doesn't you can add back the capacity parameter when calling charge_needed() to over-ride the calculation.

FencingFanatic, 2024-11-02:
The update seems to work as expected with no obvious faults. Thanks.

closed 2024-11-05 by TonyM1958