TonyM1958/FoxESS-Cloud #21: Error Running Battery Check

theslowminded, 2025-03-02:
When I try to run the 'f.battery_info(plot=2)' cell, I get the following error message:

** get_batteries(): battery 1 status not available

----------------------- BMS 1 -----------------------
SN 60***87, ECS2900-2, Version 1.010 (BMS)
SN 60***87, ECS2900-2, Version 1.10 (Battery 1)
SN 60***60, ECS2900-2, Version 1.10 (Battery 2)

---------------------------------------------------------------------------

KeyError                                  Traceback (most recent call last)

<ipython-input-8-b99a82f85b6a> in <cell line: 0>()
----> 1 f.battery_info(plot=2)

1 frames

/usr/local/lib/python3.11/dist-packages/foxesscloud/foxesscloud.py in battery_info(log, plot, rated, count, info, bat)
   3634     bat_power = bat['power']
   3635     bms_temperature = bat['temperature']
-> 3636     capacity = bat['capacity']
   3637     cell_volts = get_cell_volts()
   3638     if cell_volts is None:

KeyError: 'capacity'

TonyM1958, 2025-03-02:
I'll have a look at the code and check the error.

Please can you tell me what inverter you have, as not all models support access to battery info via the API.

theslowminded, 2025-03-02:
I have a H1-5.0-E-G2 inverter with an ECS2900-2 battery.

TonyM1958, 2025-03-03:
> I get he same error - I have a h1-6.0-e-g2 with 2 ep11's

Sorry, I haven't tested with EP11 and don't know what data to expect to get back. I can try and guide you in having a look around?

theslowminded, 2025-03-03:
The battery serial numbers were redacted by me, with '***' in the middle, as I realised this was a public page. They are being reported correctly when running the code. I validated them against what shows in FOX ESS Cloud. I have raised a WorkOrder today for Fox to upgrade the firmware on my Inverter, Battery and Datalogger as they are all not the current version. I will try running this again once that's been completed and see if that makes any difference. Otherwise, I'm happy to work with you on whatever you need.

f.get_ui gave the following:

{'protocol': 'K105xx',
 'dictionary': {'grid_code': 1},
 'parameters': [{'disable': False,
   'name': 'SystemTime',
   'block': True,
   'type': '',
   'export': '',
   'key': 'k105__system_time',
   'tips': '',
   'properties': [{'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'year',
     'range': {'hi': 2099, 'lo': 2000, 'enable': True},
     'export': '',
     'key': 'k105__system_time__year',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'month',
     'range': {'hi': 12, 'lo': 1, 'enable': True},
     'export': '',
     'key': 'k105__system_time__month',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'day',
     'range': {'hi': 31, 'lo': 1, 'enable': True},
     'export': '',
     'key': 'k105__system_time__day',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'hour',
     'range': {'hi': 23, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k105__system_time__hour',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'minute',
     'range': {'hi': 59, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k105__system_time__minute',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'second',
     'range': {'hi': 59, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k105__system_time__second',
     'tips': ''}],
   'empty': False},
  {'disable': False,
   'name': 'BasicParameters1',
   'block': False,
   'type': '',
   'export': '',
   'key': 'k105__grid_code',
   'tips': '',
   'properties': [{'elemType': {'valueType': 'string',
[... 2250 lines cut ...]
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '%',
     'level': 0,
     'name': 'Time8FDSOC',
     'range': {'hi': 100, 'lo': 10, 'enable': True},
     'export': '',
     'key': 'k105__segmented_time_mode1__time8_fd_soc',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': 'W',
     'level': 0,
     'name': 'Time8FDPwr',
     'range': {'hi': 10500, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k105__segmented_time_mode1__time8_fd_pwr',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 2,
     'name': 'TBD',
     'export': '',
     'key': 'k105__segmented_time_mode1__80',
     'tips': ''}],
   'empty': False}]}

theslowminded, 2025-04-03:
I finally managed to get the firmware updated today. This morning the code worked when I tested it, but annoyingly it’s stopped again this evening after they installed more updates. I have run the f.get_ui command again, and the protocol has changed now from K105xx to K110xx. Not sure if this will have any more, new, helpful info. I’ll paste the results tomorrow when I’m on my laptop.

theslowminded, 2025-04-04:
{'protocol': 'K110xx',
 'dictionary': {'grid_code': 1},
 'parameters': [{'disable': False,
   'name': 'SystemTime',
   'block': True,
   'type': '',
   'export': '',
   'key': 'k110__system_time',
   'tips': '',
   'properties': [{'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'year',
     'range': {'hi': 2099, 'lo': 2000, 'enable': True},
     'export': '',
     'key': 'k110__system_time__year',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'month',
     'range': {'hi': 12, 'lo': 1, 'enable': True},
     'export': '',
     'key': 'k110__system_time__month',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'day',
     'range': {'hi': 31, 'lo': 1, 'enable': True},
     'export': '',
     'key': 'k110__system_time__day',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'hour',
     'range': {'hi': 23, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k110__system_time__hour',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'minute',
     'range': {'hi': 59, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k110__system_time__minute',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 0,
     'name': 'second',
     'range': {'hi': 59, 'lo': 0, 'enable': True},
     'export': '',
     'key': 'k110__system_time__second',
     'tips': ''}],
   'empty': False},
  {'disable': False,
   'name': 'BasicParameters1',
   'block': False,
   'type': '',
   'export': '',
   'key': 'k110__grid_code',
   'tips': '',
   'properties': [{'elemType': {'valueType': 'string',
      'uiType': 'select',
      'uiItems': ['G98_UK',
       'G99_UK',
       'EN50549_NL',
[... 2649 lines cut ...]
       'EndOfDetection',
       'DetectionFailure']},
     'unit': '',
     'level': 1,
     'name': 'CT1/Meter1DetectionActive',
     'export': '',
     'key': 'k110__21__02',
     'tips': ''},
    {'elemType': {'valueType': 'string',
      'uiType': 'select',
      'uiItems': ['NotReversed', 'Reversed', 'UnknownStatus']},
     'unit': '',
     'level': 1,
     'name': 'CT1/Meter1DetectionResult',
     'export': '',
     'key': 'k110__21__03',
     'tips': ''},
    {'elemType': {'valueType': 'uint16', 'uiType': 'input'},
     'unit': '',
     'level': 2,
     'name': 'TBD',
     'export': '',
     'key': 'k110__21__04',
     'tips': ''}],
   'empty': False}]}

theslowminded, 2025-04-06:
I keep seeing the same strange behaviour. I can run the script in the morning, but it fails when I run it in the evening. Is there a newer version of the Daily.ipynb that uses my FOX API rather than my Username and Password?

TonyM1958, 2025-04-06:
Hi,

The first problem you posted is about battery_info but the latest one is about daily.ipynb. I'm not sure which script you are seeing the problem running as daily.ipynb doesn't need to call battery_info?

Please can you also post the version number that comes up.when the code signs on?

You can switch between using the Fox API and Open API by changing the import from foxesscloud.foxesscloud to foxesscloud.openapi and replacing f.username and f.password with f.api_key.

theslowminded, 2025-04-06:
Hi,

Apologies, you're absolutely right. I'd confused myself, as I was getting the same error message when running f.charge_needed. Changing the option to API seems to have resolved the issue trying to run f.charge_needed, but still having issues with the battery info. I tested all three calls against both Cloud and API, results below:

FoxESS-Cloud version 1.9.3

**f.battery_info(plot=2)**

** get_batteries(): battery 1 status not available

----------------------- BMS 1 -----------------------
SN 60JM29203BGF087, ECS2900-2, Version 2.004 (BMS)
SN 60JM29203BGF087, ECS2900-2, Version 1.14 (Battery 1)
SN 60JS2920328C260, ECS2900-2, Version 1.14 (Battery 2)

** get_named_settings(): BatteryVolt was not recognised

**f.get_batteries()**

** get_batteries(): battery 1 status not available

[{'soc': 86,
  'residual': 0.0,
  'soh': '99',
  'ratedCapacity': 5760.0,
  'current': -2.9,
  'descUrl': '',
  'ratedCapacityDisplay': True,
  'volt': 119.3,
  'statusName': 'Discharge',
  'temperature': 29.7,
  'warning': False,
  'power': 0.319,
  'status': 2,
  'info': {'masterSN': '60JM29203BGF087',
   'masterBatType': 'ECS2900-2',
   'masterVersion': '2.004',
   'slaveBatteries': [{'sn': '60JM29203BGF087',
     'batType': 'ECS2900-2',
     'version': '1.14'},
    {'sn': '60JS2920328C260', 'batType': 'ECS2900-2', 'version': '1.14'}]}}]

**f.get_battery()**

** get_battery(): battery status not available

OR

FoxESS-Cloud Open API version 2.8.1

**f.battery_info(plot=2)**

----------------------- BMS 1 -----------------------
** get_cell_volts(): not available via Open API

**f.get_batteries()**

[{'soc': 85.0,
  'volt': 118.1,
  'current': 2.0,
  'power': 0.22,
  'temperature': 29.6,
  'residual': 0.057,
  'residual_handling': 1,
  'soh': None,
  'soh_supported': False,
  'status': 1,
  'count': None,
  'ratedCapacity': None,
  'capacity': 0.067,
  'charge_rate': 48.0,
  'charge_loss': 0.974,
  'discharge_loss': 0.974}]

**f.get_battery()**

{'soc': 85.0,
 'volt': 118.0,
 'current': 4.4,
 'power': 0.501,
 'temperature': 29.6,
 'residual': 0.057,
 'residual_handling': 1,
 'soh': None,
 'soh_supported': False,
 'status': 1,
 'count': None,
 'ratedCapacity': None,
 'capacity': 0.067,
 'charge_rate': 48.0,
 'charge_loss': 0.974,
 'discharge_loss': 0.974}

TonyM1958, 2025-04-07:
Thanks for the update, that makes sense. The full battery information simply isn't available via Open API so the responses there are as expected. However charge_needed should work via either, although setting schedules via Open API is also unreliable. This is why I recommend using the Fox API (until Fox fix their Open API).

Whether Fox returns the battery status correctly does seem to be hit and miss. A status of 0 means the battery is offline and unable to report current SoC, while a status of 1 means the battery is online and can report current SoC. The problem is that the Fox API seems to return offline at times. You also see this in the Fox app, where the Devices page randomly reports the battery is offline (also reported by various people in the FB group).

When you get this error, please can you add f.battery to a cell and run this. This should show the battery details that the Fox API is returning for your battery and will help with further diagnosis. I can then look at the best way we can work around any inconsistencies.

theslowminded, 2025-04-07:
Ran charge_needed over Fox API and got the error, so ran f.battery as requested. You only mention states 0 and 1 above, what does 2 mean that's been returned below? Can you also explain what the first line "Charge time: hh:mm-hh:mm" means? The reason I ask is that it always says a different time period. If it's the time being used to set a force charge schedule, should this not say between 02:00 and 05:00 for Flux? Do I need to run f.set_tariff('flux', region='H') each time I run charge_needed, or is it a one-time assignment to set the tariff?

FoxESS-Cloud version 1.9.3

f.charge_needed(update_settings=0, timed_mode=2, force_charge=0)

---------------- charge_needed ----------------
Charge time: 20:00-23:00
** get_battery(): battery status not available

f.battery

{'status': 2,
 'soc': 100,
 'volt': 119.8,
 'current': 1.2,
 'power': 0.147,
 'residual': 57.300000000000004,
 'temperature': 29.1,
 'timestamp': '',
 'warning': False,
 'descUrl': '',
 'info': {'masterSN': '60JM29203BGF087',
  'masterBatType': 'ECS2900-2',
  'masterVersion': '2.004',
  'slaveBatteries': [{'sn': '60JM29203BGF087',
    'batType': 'ECS2900-2',
    'version': '1.14'},
   {'sn': '60JS2920328C260', 'batType': 'ECS2900-2', 'version': '1.14'}]},
 'count': 2,
 'residual_handling': 1,
 'soh': None,
 'soh_supported': False}

TonyM1958, 2025-04-08:
Thanks, that helps. Fox doesn't provide any decoded for the status codes and I don't think I've seen 2 before. Please can you check what the status of the battery shows in the app? (Just tap Devices).

The current currently checks is status == 1 for online, but it might be better if I simply check for status != 0 instead.

Also interesting that the residual is 57.3. The scaling of Residual seems to vary with battery type. Given your SoC is 100%, this must mean the battery residual / capacity is 5.73kWh. I may also need to look at the scaling. If charge_needed shows the wrong capacity, you can over-ride this by adding the parameter capacity=5.73 to the call.

theslowminded, 2025-04-08:
The battery shows as Offline Discharging in the app, but allows me to select it and look at the Analysis Graph. All looks fine in the web cloud portal

theslowminded, 2025-04-09:
No problem at all. Enjoy your time off! I can wait.

One slight development today. When I checked the app, the battery showed as Online all the time it was being charged from solar (I forgot to run f.battery, as I didn't think of it at the time. Will try tomorrow if it's the same). As soon as the sun went down and I started pulling from the battery, it changed again to Offline Discharging. I've raised a work order with Fox to tell them there's an issue with the way the app is reporting.

charge_needed works today, even with the app showing the battery as offline, so your change seems to be working. Both the Capacity and Residual were incorrect with a partially depleted battery. After manually supplying the capacity=5.73 value, the residual calculated correctly in the next run.

How do I get the charge schedule to use the 02:00 - 05:00 Flux low-rate window? I've run 'f.set_tariff('flux', region='H')' but it still wants to charge between 00:00 and 01:00

Charge needed: 2.88kWh (00:40)
  SoC now:     72% at 21:16 on 2025-04-09
  Start SoC:   44% at 00:00 (2.10kWh)
  Charge:      00:00-01:00 (3.04kWh added)
  End SoC:     100% at 01:00 (4.73kWh)
  Contingency: 24% SoC (1.12kWh)

TonyM1958, 2025-04-10:
I've looked through the code and setting the tariff should work. You can check the settings by inspecting this variable (type this in a cell and run it):

```
f.tariff
```

Please have a look and check the tariff is set correctly? You need to run set_tariff() after any reset of the Kernel and before you run charge_needed().

Chrisbrown01942, 2025-05-21:
Hi Tony, I have an issue with the battery check since updating my inverter to the latest firmware Master 1.67, Slave 1.02, ARM 1.76, error is battery volt not recognised for ECS2900 on the latest firmware too. Regards, Chris

Chrisbrown01942, 2025-05-21:
[{'protocol' 'H118xx',.txt](https://github.com/user-attachments/files/20374307/protocol.H118xx.txt)

So, I got this back. Understand none of it, hopefully, it's what you need. Yeah, I only have an end-user account.

TonyM1958, 2025-05-21:
I've pushed out an updated library that recognises the H118 protocol but assumes the cell voltage and temperature key positions have stayed the same. This may or may not work.

The updated library should sign on as version 1.9.6 after the PyPi servers have replicated the files in a few minutes. Given it a go and see what results / errors this gives you. If the keys aren't right, we may need to find someone with installer access to H1 with Manager 1.76 that can run get_ui() to see where the keys have moved to.

Chrisbrown01942, 2025-05-22:
ThanksTony, that did. All is working as before firmware update, many thanks