TonyM1958/FoxESS-Cloud #22: pv_daily.ipynb error

epic999, 2025-03-02:
Init of notebook works okay,

# list of dates to use. today=0 will get yesterdays complete data, today=1 will get todays partial data, or enter a specific date in 'YYYY-MM-DD' format
d = f.date_list(today=0)

# get the data from Fox Cloud
result = f.get_history('day', d=d, v=['pv1Power', 'pv2Power'], plot=1)

# print the results
for r in result:
  print(f"{r['date']} / {r['name'].replace('Power', '')}: Peak output was {r['max']:.2f} kW at {r['max_time']}. Total energy generated was {r['kwh']:.1f} kWh")

** get_device(), no list result data, errno = 41808: Token is invalid
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
[<ipython-input-2-c04dcc3b8d2d>](https://localhost:8080/#) in <cell line: 0>()
      6 
      7 # print the results
----> 8 for r in result:
      9   print(f"{r['date']} / {r['name'].replace('Power', '')}: Peak output was {r['max']:.2f} kW at {r['max_time']}. Total energy generated was {r['kwh']:.1f} kWh")

TypeError: 'NoneType' object is not iterable

TonyM1958, 2025-03-02:
Hi. The first error here is the problem, where it says Token is invalid. That looks like a problem with the API token being used so the code is unable to get the details of your inverter. Please can you check your API is being entered correctly in the first cell using double quotes like this and give it another go?

```
f.api_key="<your api key>"
```

epic999, 2025-03-03:
had a missing hyphen,

thanks

closed 2025-03-03 by TonyM1958