TonyM1958/FoxESS-Cloud #23: Battery capacity or residual are reported incorrectly

TonyM1958, 2025-03-11:
The way that Fox ESS Cloud reports the battery capacity and residual can vary with some inverters and BMS firmware. There are 2 problems:

Firstly, the Residual scaling may be incorrect. Normally, the value returned is in units of 10Wh. However, in some cases, the actual value is in kWh (even though the units still say it is Wh). To adjust for this, set this variable during initialisation:

```
f.residual_scale = 1
```

The default value is 0.01 to scale from 10Wh to 1kWh. Changing the value to 1 will set the scale to 1kWh by default.

The second issue is the way that the value that the BMS reports for Residual. For older versions of BMS firmware (< 1.014), Residual is the current amount of energy stored in the battery at the current SoC. When you update the firmware, the BMS reports the current battery capacity instead of the Residual. To allow for this, the code includes a setting for how to work out battery capacity and residual:

```
f.residual_handling = 1    # Residual returns current battery energy
f.residual_handling = 2    # Residual returns current total capacity (HV / ECS)
f.residual_handling = 3    # Residual returns current capacity per battery (Mira)
```

You can test the battery information being returned:

```
f.get_battery()
```

gavhughes87, 2025-03-12:
Hi Tony
i have added the following into the first cell to initialise the notebook (daily.ipynb) but it doesn't return any data. im assuming i have added it incorectly as i am a complete novice at this. i have added this code below after the PV output credentials. Can you send an example of how the cell should look after adding in the residual handling above.
thanks

# Residual scaling
f.residual_scale = 1
f.residual_handling = 1    # Residual returns current battery energy
f.residual_handling = 2    # Residual returns current total capacity (HV / ECS)
f.residual_handling = 3    # Residual returns current capacity per battery (Mira

TonyM1958, 2025-03-12:
Hi, Let me know what batteries and BMS firmware you have?

You just need one of the lines that set residual_handling, depending on which firmware you have. For example, try this to start with:

```
f.residual_scale = 1
f.residual_handling = 1
```

If you then run charge_needed(), it should show what it's getting for the battery info. Post this and we can see if any further adjustments are needed.

gavhughes87, 2025-03-12:
Hi Tony, I have added the above into the first cell and ran the notebook. i ran the next step f.get_battery() which returned this. residual and capacity still showing incorrect.

{'status': 1,
 'soc': 85,
 'volt': 184.4,
 'current': -8.4,
 'power': -1.511,
 'residual': 0.075,
 'temperature': 19.4,
 'timestamp': '',
 'warning': False,
 'descUrl': '',
 'info': {'masterSN': '60JM29203BNF105',
  'masterBatType': 'ECS2900-2',
  'masterVersion': '1.011',
  'slaveBatteries': [{'sn': '60JM29203BNF105',
    'batType': 'ECS2900-2',
    'version': '1.14'},
   {'sn': '608S2920314J269', 'batType': 'ECS2900-2', 'version': '1.14'},
   {'sn': '608S2920315J034', 'batType': 'ECS2900-2', 'version': '1.14'}]},
 'count': 3,
 'residual_handling': 1,
 'soh': None,
 'soh_supported': False,
 'capacity': 0.088,
 'charge_rate': 14.4,
 'charge_loss': 0.974,
 'discharge_loss': 0.974}

TonyM1958, 2025-03-12:
Thanks - the Fox API is returning the Residual in Wh for my HV battery but in kWh for your battery by the look of it.

I've updated the code to try to fix this. If you restart your notebook (Runtime, Disconnect and Delete) and then run the cells again and see what the results are. The initialisation should sign-on as v1.9.1 if you are running the updated code.

gavhughes87, 2025-03-12:
Hi Tony, the initialisation came back as FoxESS-Cloud version 1.9.2 and when i ran the { f.get_battery() } it returned: { ** get_battery(): battery status not available }. same for the charge needed code. My battery SOC is currenyly 99%

TonyM1958, 2025-03-12:
> Hi Tony, the initialisation came back as FoxESS-Cloud version 1.9.2 and when i ran the { f.get_battery() } it returned: { ** get_battery(): battery status not available }. same for the charge needed code. My battery SOC is currenyly 99%

status was 1 when you posted the results previously and I don't think I've changed anything around this. It looks like Fox is saying your battery is offline. Please can you have a look under Devices in Fox App v2 and see if it also says the battery is online or offline?

There are known problems with Fox showing the battery is offline when it is working so I'm wondering if this is another symptom of them not returnning the battery status correctly?

gavhughes87, 2025-03-12:
Hi Tony, the initialisation came back as FoxESS-Cloud version 1.9.3 . this time it worked as expected. see results below: 
---------------- charge_needed ----------------

Battery Info:
  Capacity:    8.65kWh
  Residual:    8.56kWh
  Voltage:     188.2V
  Current:     -5.8A
  State:       Charging (1.090kW)
  Min SoC:     10% (0.86kWh)
  Current SoC: 99%
  Max SoC:     100% (8.65kWh)
  Max Charge:  14.4A
  Temperature: 19.4°C
  Resistance:  0.26 ohms
  Nominal OCV: 183.1V at 60% SoC

Device Info:
  Model:     KH7
  Rating:    7.00kW
  Export:    7.00kW
  Charge:    14.4A, 2.69kW, 96.3% efficient
  Discharge: 50.0A, 7.22kW, 97.0% efficient
  Inverter:  101W power consumption
  BMS:       50W power consumption
  Work Mode: SelfUse

Consumption (kWh):
  2025-03-09 11.1, 2025-03-10  8.3, 2025-03-11  7.3
  Average of last 3 days: 8.9kWh

Generation (kWh):
  2025-03-09 13.2, 2025-03-10  3.7, 2025-03-11 10.6
  Average of last 3 days: 9.2kWh

Charge needed: 1.46kWh (00:33)
  SoC now:     99% at 15:39 on 2025-03-12
  Start SoC:   25% at 06:00 (2.20kWh)
  Charge:      06:00-07:00 (2.84kWh added)
  End SoC:     56% at 07:00 (4.82kWh)
  Contingency: 26% SoC (2.27kWh)
  PV cover:    103% (9.2/8.9)

Configuring schedule:
   06:00-06:59 ForceCharge, minsoc 10%, maxsoc 100%

No changes made to charge settings.

thanks for your help on this one, really appreciate it.