TonyM1958/FoxESS-Cloud #24: Max SOC not being set in schedule

AdamJacobs666, 2025-07-13:
I am trying to set a charging schedule, first by using the set_period method to create a period object, and then passing that period to the set_schedule method.

This is almost successful, but it seems that if I pass a max_soc parameter to the set_period method, it is ignored, and the max SOC gets set to 100%.

I have found a workaround by manually creating a 'maxSoc' key in the resulting period object (which seems to be just a dictionary) with the required value before passing it to the set_schedule method, and then everything works as intended. But if you could fix the set_period method (or let me know what I have misunderstood in the documentation), that would be great.

BTW, thanks ever so much for creating this software. As the proud owner of a new FoxESS solar system, this has made it way more easy to interact with it than it would have been otherwise. Keep up the good work!

TonyM1958, 2025-07-13:
Hi,

The code has to work out whether your inverter supports setting max soc in schedules as sending the parameter if it isn't supported returns an error. Unfortunately, Fox does not provide a flag to indicate if max soc is supported or not.

If you call f.get_schedule before you try to set a schedule, the code will load the result and check if max soc is present. If it is, then the inverter supports max soc and the code sets a flag that allows max soc to be set when you send a schedule. You can check this flag by inspecting f.schedule and looking for maxosc == True.

So, all you should need to do is to call f.get_schedule once before you call set_period or set_schedule and it should work. Or, you can bodge is by forcing f.schedule['maxsoc'] to True.

Please let me know if this doesn't fix the issue and I will take another look.

AdamJacobs666, 2025-07-14:
Thanks for the prompt reply. Yes, calling f.get_schedule first does indeed fix the problem. If I do that then everything works as expected.

Thank you!

You might want to add a note about that to the documentation for the set_period method. I can see that this is documented under get_schedule, but it's easy to miss that if you're just trying to set the period.

TonyM1958, 2025-07-14:
Thanks - README updated. It's set_period() that decides whether to include max soc rather than set_schedule() so I've moved the notes to make it clearer.

closed 2025-07-14 by TonyM1958